dubstar-04/TurningAddon #4: PathJobGui.ERROR: 'Direction'

daveseff, 2020-10-13:
Getting This error when selecting new Job from Path workbench:

`
PathJobGui.ERROR: 'Direction'
Traceback (most recent call last):
  File "/usr/lib/freecad/Mod/Path/PathScripts/PathJobGui.py", line 1352, in Create
    obj.ViewObject.Proxy.editObject(obj.Stock)
  File "/usr/lib/freecad/Mod/Path/PathScripts/PathJobGui.py", line 199, in editObject
    return self.openTaskPanel('Stock')
  File "/usr/lib/freecad/Mod/Path/PathScripts/PathJobGui.py", line 181, in openTaskPanel
    self.taskPanel.setupUi(activate)
  File "/usr/lib/freecad/Mod/Path/PathScripts/PathJobGui.py", line 1235, in setupUi
    self.setupOps.setupUi()
  File "/usr/lib/freecad/Mod/Path/PathScripts/PathSetupSheetGui.py", line 281, in setupUi
    op.setupUi()
  File "/usr/lib/freecad/Mod/Path/PathScripts/PathSetupSheetGui.py", line 174, in setupUi
    prop = self.prototype.getProperty(name)
  File "/usr/lib/freecad/Mod/Path/PathScripts/PathSetupSheetOpPrototype.py", line 200, in getProperty
    return self.properties[name]
KeyError: 'Direction'
`

This is against the latest git version (g68ad4f2f1d) of FreeCAD

dubstar-04, 2020-10-13:
Hi @daveseff, 

Thanks for the bug report.

The error you have pasted doesn't seem to be related to the Turning Addon but an error in the core FreeCAD Path workbench. 

Would you mind posting on the FreeCAD forum: https://forum.freecadweb.org/viewforum.php?f=15&sid=5530dda4e7f7a04b3f5d7e0604aa0ea4

Thanks, 

Dan

dubstar-04, 2020-10-13:
@daveseff I have opened a forum post here: 

https://forum.freecadweb.org/viewtopic.php?f=15&t=51111 

Closing this issue.

dubstar-04, 2020-10-22:
@daveseff please accept my apologies. It does appear that this is caused by the add-on.

daveseff, 2020-10-23:
No problems, mate. I'm having a look through the code when I get free time to try to help out.

dubstar-04, 2020-10-23:
@daveseff I will be changing how liblathe fundamentally works over the next few weeks. The current aim is to define how tools are created in FreeCAD and passed to the library. I think I have it all worked out. We are not actually generating usable gcode yet but I hope for that to change shortly.

closed 2020-10-25 by dubstar-04

dubstar-04, 2020-10-26:
Hi @daveseff I had a look over the code and I think this issue is now fixed. If you are interested in lathe functionalility I would appreciate you testing the add-on periodically. It doesn't currently generate valid tool paths but that should start to change over the next couple of weeks. 

Current aim is to get a first release of liblathe that includes, profiling, facing and parting, then start working on the FreeCAD UI. 

Thanks again, 

Dan

daveseff, 2020-10-27:
Thanks, Will do.

daveseff, 2020-10-27:
Looks Better. Cheers,

commit c885a58b (2020-10-25): update operation properties. fixes #4